badgeteam/mch2022-firmware-esp32 #4: Camp wifi credentials need to be added as the default setting

renzenicolai, 2022-05-07:


robotman2412, 2022-05-09:
I've contacted the NOC and it seems that we need to use the `MCH2022-legacy` network.
This is because the ESP32-WROVER-E doesn't support 5GHz WiFi (the recommended type).
The network will be encrypted using `WPA2 802.1X`, using any username and password (they don't check those).

robotman2412, 2022-05-09:
See also: https://wiki.mch2022.org/Network

renzenicolai, 2022-05-13:
Cool so we need enterprise WPA support :smile: , that's going to be fun

robotman2412, 2022-05-28:
With the enterprise WPA2 enterprise support almost finished, i think we need to talk about storing networks.
The firmware has references to a list of known networks, but currently there is no such thing.
How would we go about making and keeping track of this list?

robotman2412, 2022-06-01:
The credentials just changed:

- SSID: `MCH2022`
- username: `mch2022`
- password: `mch2022`

robotman2412, 2022-06-01:
The basic menu has been tested as well.
Could anyone verify this?

Branch:
https://github.com/badgeteam/mch2022-firmware-esp32/tree/wifi_test

I testes with FreeRadius on a NetGear R7000.

Network settings:
- 2.4GHz WPA2 enterprise
- MSCHAPv2 phase 2
- aforementioned username/password

robotman2412, 2022-06-22:
It has yet to be tested, but the latest commits on master have the new WiFi defaults, which comply with NOC requests.
This issue can be closed when the test succeeds.