badgeteam/mch2022-firmware-esp32 #43: Forward button presses to FPGA in FPGA download mode

renzenicolai, 2022-05-24:
When the FPGA bitstream has been downloaded and the ESP32 is waiting for a new start command from the PC to load a new bitstream the ESP32 should forward button presses to the FPGA as simple SPI transfers.

The button presses are sent to a queue (that can be added to the FPGA downloader function as an argument from main, have a look at the other functions of the firmware that use buttons) and the ice40 driver has a function to send data to the ICE40 via SPI: esp_err_t ice40_send(ICE40* device, const uint8_t* data, uint32_t length);

The FPGA people specified the following:

Packet would be f4 xx xx yy yy  with the  xx xx being current state and yy yy the change mask.
(the F4 byte at the start is a command byte / constant)

FransFaase, 2022-05-26:
The above commits seem to implement the required function. I tested this with Buttons.bin.

Bits are mapped to the following keys:
 0 - joystick down
 1 - joystick up
 2 - joystick left
 3 - joystick right
 4 - joystick press
 5 - home
 6 - menu
 7 - select
 8 - start
 9 - accept
10 - back

FransFaase, 2022-05-26:
Did some code clean-up. Button processing has now be moved to fpga_process_events. I decided on this generic name, because maybe also other events will be send to the FPGA in the future using this function.

The commit also includes a version of the fpga.py which keeps the port open and echos messages from USB serial port about the progress of the download or possible error messages. This could aid debugging.