badgeteam/mch2022-firmware-esp32 #131: Uninstaller hangs

Pwuts, 2022-07-11:
### What?
Uninstalling ice40_doom from internal storage hangs. The files `metadata.json` and the two `.dat` files are still there after waiting a long time and power cycling. (Checked with built-in file browser)

### Steps to reproduce:
1. Install ice40_doom
2. Uninstall ice40_doom

renzenicolai, 2022-07-12:
Please test this again by compiling the firmware, running "make flash monitor" and starting the uninstall process.
If the badge crashes please provide the backtrace as printed by the monitor tool.

a2800276, 2022-07-15:
I uninstalled a different app (basically just the compiled template app) and the unintstaller also crash. I uninstalled previously with no issues, so maybe this only affects the subsequent uninstalls  ...Anyway, here's the trace:

I assuem it's the saem issue. Firmware 1.1 from OTA

```
W (4030) Metadata: Directory not found: /sd/apps/esp32
W (4040) Metadata: Directory not found: /internal/apps/ice40
W (4040) Metadata: Directory not found: /sd/apps/ice40
W (4070) Metadata: Directory not found: /sd/apps/python
Removing AppFS entry...
Removing directory '/internal/apps/esp32/timtest'...
Removed a file: /internal/apps/esp32/timtest/README.md
Removed a file: /internal/apps/esp32/timtest/icon.png
Removed a file: /internal/apps/esp32/timtest/icon.py

assert failed: block_locate_free heap_tlsf.c:447 (block_size(block) >= size)

Backtrace:0x40082026:0x3ffbc0100x4008c929:0x3ffbc030 0x40094705:0x3ffbc050 0x40092476:0x3ffbc170 0x40092f45:0x3ffbc190 0x400821cd:0x3ffbc1b0 0x40082231:0x3ffbc1d0 0x4009477c:0x3ffbc1f0 0x400947a0:0x3ffbc210 0x400da581:0x3ffbc230 0x400dc36e:0x3ffbc2d0 0x400dc7a0:0x3ffbc300 0x400dc963:0x3ffbc3c0 0x400db794:0x3ffbc470 0x400d973e:0x3ffbc780 0x401a0037:0x3ffbc800 0x4009013d:0x3ffbc820 
0x40082026: heap_caps_malloc_base at /home/a2800276/projects/mch/testapp/build/../esp-idf/components/heap/heap_caps.c:147

0x4008c929: xTaskIncrementTick at /home/a2800276/projects/mch/testapp/build/../esp-idf/components/freertos/tasks.c:3156

0x40094705: hal_mac_tx_is_cbw40 at ??:?

0x40092476: __assert_func at /home/a2800276/projects/mch/testapp/build/../esp-idf/components/newlib/assert.c:62

0x40092f45: ppSearchTxframe at ??:?

0x400821cd: heap_caps_realloc_base at /home/a2800276/projects/mch/testapp/build/../esp-idf/components/heap/heap_caps.c:409

0x40082231: heap_caps_realloc at /home/a2800276/projects/mch/testapp/build/../esp-idf/components/heap/heap_caps.c:430

0x4009477c: wdev_bank_load at ??:?

0x400947a0: wDev_GetNoiseFloor at ??:?

0x400da581: pax_join at /home/a2800276/projects/mch/testapp/build/../components/pax-graphics/src/helpers/pax_mcr.c:169 (discriminator 1)

0x400dc36e: pax_rect_shaded at /home/a2800276/projects/mch/testapp/build/../components/pax-graphics/src/helpers/pax_dh_shaded.c:365

0x400dc7a0: pax_rect_shaded at /home/a2800276/projects/mch/testapp/build/../components/pax-graphics/src/helpers/pax_dh_shaded.c:437

0x400dc963: pax_rect_shaded at /home/a2800276/projects/mch/testapp/build/../components/pax-graphics/src/helpers/pax_dh_shaded.c:482

0x400db794: paxmcr_rect_shaded at /home/a2800276/projects/mch/testapp/build/../components/pax-graphics/src/helpers/pax_dh_mcr_shaded.c:395 (discriminator 1)

0x400d973e: paxmcr_tri_shaded at /home/a2800276/projects/mch/testapp/build/../components/pax-graphics/src/helpers/pax_dh_mcr_shaded.c:96

0x4009013d: insert_free_block at /home/a2800276/projects/mch/testapp/build/../esp-idf/components/heap/heap_tlsf.c:253
 (inlined by) block_insert at /home/a2800276/projects/mch/testapp/build/../esp-idf/components/heap/heap_tlsf.c:269
 (inlined by) tlsf_add_pool at /home/a2800276/projects/mch/testapp/build/../esp-idf/components/heap/heap_tlsf.c:689

ELF file SHA256: 375d83616ff9681b
```

closed 2022-07-23 by smunaut