badgeteam/mch2022-firmware-esp32 #137: Inconsistency between error message and app name for Python

smunaut, 2022-07-23:
If python isn't available, the error message refers to it as "BadgePython", but in the Hatchery it's named "Python" and this confused some people so it'd be better to make them match.

Question is do we change the error message to say "Python", or update the Hatchery to say "BadgePython" ?

@renzenicolai opinion ?

renzenicolai, 2022-07-23:
Change to python for now and eventually change back once hatchery can do that

a2800276, 2022-07-23:
There's also "Micropython" ... I think we need to make sure we are more clear when documenting...

renzenicolai, 2023-04-13:
Some clarification:

MicroPython is the upstream project. BadgePython is the name of our project. BadgePython is an extension of MicroPython adding all kinds of badge related features such as a graphics stack and drivers/abstractions for most of the peripherals found on the supported badges.

On the hatchery apps currently can't have a title different from their slug name, the slug for the BadgePython app for the MCH2022 badge is "python" as it's the default python implementation for the MCH2022 badge. Maybe a future version of the hatchery will fix this 😄 

For now the error message has been changed to "Python is not installed\n\nPlease install 'Python'\nusing the Hatchery under\n'ESP32 native binaries\\Utility".

closed 2023-04-13 by renzenicolai