badgeteam/mch2022-firmware-esp32 #142: hdmi pmod demo app src?

xeniter, 2022-08-11:
hi,

didn't find infos/src to the hdmi demo pmod app?
Is it online somewhere?

see https://mch2022.badge.team/projects/hdmi_demo
have a pmod with the TI TFP410 would be interesting if it works with that

smunaut, 2022-08-11:
The ESP32 app source is https://github.com/badgeteam/mch2022-app-hdmi
And the FPGA source is https://github.com/smunaut/mch2022-ice40/tree/master/projects/hdmi
The board design it uses is https://github.com/smunaut/DVI-PMOD

There is no encoder on board, just direct connection from PMOD to HDMI data lines and the TMDS encoding is done on the FPGA itself.

What PMOD do you have ?

xeniter, 2022-08-11:
big thx, this one: 

BML 3bit DVI over single-PMOD
https://blackmesalabs.wordpress.com/2017/12/15/bml-hdmi-video-for-fpgas-over-pmod/

closed 2022-08-11 by xeniter