badgeteam/mch2022-firmware-esp32 #146: FSLS times out for large folders

Pwuts, 2023-04-04:
The FSLS command seems to either fail or respond very slowly when listing large folders. The more items in a folder, the higher the chance of a time-out.

Experienced/tested with https://mch2022-badge-webusb.vercel.app. Try `/internal/apps/python/citycontrol` on the MCH2022 badge for example.

renzenicolai, 2023-04-04:
Interesting, try to dump all communication into the console, I bet the badge spits out some ascii text with a crashlog if it stops responding...

I will ask details via Tg.

Pwuts, 2023-04-04:
This is a good use case for a "dump everything" switch in the settings :)

Pwuts, 2023-04-04:
Looks like this is an issue with the user agent. Works fine with devtools closed, starts faltering with devtools open. Closing for now.

closed 2023-04-04 by Pwuts